Photonsters/PhotonFileEditor #1: We need walktrough on how to install/use this

X3msnake, 2018-06-16:


NardJ, 2018-06-16:
I have added it to the README.md.  I have just finished implementing editing of all data field in the Photon file. The GUI is not very pretty at this moment, but it works well enough. The next functionality to implement will probably be importing/replacing of bitmap data. Any feedback of any kind is more than welcome!

NardJ, 2018-06-16:
Another thing: there are a number of data-fields in the Photon file for which I do not know the usage (neither is it available in the [010-template](https%3A%2F%2Fgithub.com%2FaltLab%2Fphoton-resin-calibration%2Fblob%2Fmaster%2F010editor-template%2Fanycubic-photon-hextemplate.bt)). Any more info about these fields would help me!

X3msnake, 2018-06-16:
@NardJ does this help?

https://github.com/Reonarudo/pcb2photon/issues/2

X3msnake, 2018-06-17:
@NardJ Excelent work with the walk-through i'll be testing it this weekend and will post back ;)

NardJ, 2018-06-18:
That helped, I now have previews available!

closed 2018-06-19 by X3msnake