Photonsters/PhotonFileEditor #3: Can't open File

X3msnake, 2018-06-17:
[image]

NardJ, 2018-06-18:
The file open dialog is not very stable yet especially after choosing Cancel before. I tried reproducing your error by downloaded the zip from github and starting it from the command prompt. I did not manage te get this error. Do you know which action you took since starting the Photon Editor up until this error?
Also did you try again after restarting the Photon Editor?
Btw: Thx for the screenshot with the CMD-window open.

NardJ, 2018-06-18:
I found the bug which occured when selecting a file to open after previously Canceling on the FileDialog. Uploaded new files. This could also solve your problem...

X3msnake, 2018-06-18:
@NardJ ok thanks will test later tonight

NardJ, 2018-06-18:
I think I have a solution. I made some static methods in a class, without explicitely defining them as statics. I will upload a new version, to test.

X3msnake, 2018-06-18:
@NardJ ok i'll wait for your files, i was trying to see if by tinkering I could fix it but my python-fu is lame :P

X3msnake, 2018-06-18:
[image]

NardJ, 2018-06-18:
Ok, I have tested this on all pc's I have at home: Windows 10 32-bit tablet, Windows 10 64-bit old laptop, Windows 10 64-bit older laptop, Windows 10 64-bit new laptop and Linux Mint 18.3 Cinnamon Desktop. I tested on Python installed a two year ago with PyCharm-editor and with fresh installs of  bare Python/Pygame as stated in the Readme.md.

I also used the source-code which was this afternoon on github. However none of them gave any problems/errors.

From the screenshot I see,  you are using an older Windows version, Vista? Maybe this is the problem. Do you have another machine/OS to test it on? Meanwhile, I will set up a virtual machine with Vista installed to check if I can reproduce the errors.

X3msnake, 2018-06-19:
Win7 64bit

X3msnake, 2018-06-19:
@NardJ Maybe it's anaconda, what python version are you writing on?

X3msnake, 2018-06-19:
It was the python version :D 

For windowss users
As alternative to installing python from the link you suggested users can also download anaconda 3.6
https://www.anaconda.com/download/

The rest is the same, go windows type anaconda and run anaconda prompt 

there just type:   python -m pip install -U pygame --user  to install the required library
Download the files
change to the unziped files dir and type python PhotonEditor.py 

and enjoy

[image]

NardJ, 2018-06-19:
Aha didn't think of this, glad it is solved! Will update the Readme.md.

NardJ, 2018-06-19:
Updated Readme.md

closed 2018-06-19 by NardJ