Photonsters/PhotonFileEditor #5: Unknown fields

Andoryuuta, 2018-06-18:
I seems that, at least from the readme photo, you are currently missing a few of the non-zero fields. I compared mine with the template from Leonardo that I think you are using, here are the differences:

unknown3 = Preview Header Offset
unknown4 = Layer Headers Offset
unknown5 = Preview Thumbnail Header Offset
unknown7 = LightCuring/Projection type // (1=LCD_X_MIRROR, 0=CAST)

Also, fair warning; some offsets seem to seek relatively depending on the MSB being set. In specific, I know the MSB is used for the offset within the layer image headers, and it is probably used for the the offset within the preview image headers. I have yet to actually come across this being used in any `.photon` or  `.cbddlp` files (although I've only tested very small files from both versions of the slicer(s)), nor have I been able to figure out the firmware update format to disassemble and verify its usage.

 Regardless, you might want to add a simple check like:

```Go
if(LayerHeader.ImageDataOffset&0x80000000 != 0) {
    panic("Potentially non-implemented relative seek!")
}
// and
if(PreviewHeader.ImageDataOffset&0x80000000 != 0) {
    panic("Potentially non-implemented relative seek!")
}

```

NardJ, 2018-06-18:
Thx, Unknown 7 was new to me! Do you perhaps know what unknown0 and 6 are? (Last ones missing)

Andoryuuta, 2018-06-18:
unknown0 is currently a fixed uint64 constant in the slicing software, I assume its just the file magic.
```asm
__text:000000010002EA81                 mov     rax, 112FD0019h
```
It _might_ contain version information, but the slicer's .photon/cbddlp loader (for previewing) doesn't touch it.

unknown6 isn't accessed for (preview) loading or for writing as far as I can tell,.