Photonsters/PhotonFileEditor #7: User UI improvement

X3msnake, 2018-06-19:
# Input/Output
- Allow user to set default values for the fields
- Allow user to save settings and load them from a list (resin settings)

# 
- Allow use of up/down arrow to navigate layer / frames

NardJ, 2018-06-19:
Import bitmaps is maybe already done. There is a functionality (Edit menu) which maybe is the same: All old bitmaps of the .photon fil are removed. The old setings of the first layer are applied to all new bitmaps/layers. (It is however not tested yet.) Is this what is needed?

NardJ, 2018-06-19:
Great list! There are a view items I need some extra info on. An example when and why it is needed would help me out..

1) Default values: Is this needed for new files or to set settings on existing/loadedfiles?
2) Replace image based on current layer/frame: For this one I need an example to understand it.
3) Allow up/down arrow to navigate layer / frames: Isn't this already present? See Arrows at topleft corner. What is needed/exepected?
4) Implement Slider drag: Isn't this already present? See and slider bar at right edge of layer image. What is needed/expected?

X3msnake, 2018-06-19:
@NardJ 

1.
This ties to being able to load and save settings. kind of like the optin to chose PLA settings or save new from a list on Cura. this will allow a user to open up a file and repourpouse to use with a new resing for example. 

2. 
Maybe i'm rading this wrong, but import for me reads as i have to have the same number of images to replace in the file? can you clarify how you implemented it now?

But what i was saying is; Say you don't want to replace all images on your file just the one you are looking at. It is usefull to have a replace feature, you select the image you want from file and it modify the file in context.

3.
I'm refering to keyboard keys in parallel to the ui arrows 

4.
I mean real drag-and-drop like the scrollbars in explorer, now you can click on a spot and it goes there but you can't use the bar to shuffle up/down with the mouse drag.

NardJ, 2018-06-20:
Thx this is clear now. Two answer your question regarding

2. I store the settings from layer 0, then I delete all layer images and
layer defs, iterate through all images in the given path and add them using
the storer settings from the ood layer 0.

I will implement your use case also. However while bug hunting yesterday I
found the import of images (or rather encode to B/W RLE data) is not
working correctly. Will fix this first.

Regarding

4. This will mean for quick enough access and display on mousedrag, I have
to store all decompressed images in memory too. Will look into the amount
of mem this uses.

X3msnake, 2018-06-20:
You can store low rez versions of the layers if needed it is perfectly acceptable to have a lower quality preview while browsing with a dragbar

NardJ, 2018-06-20:
Off coure, good one.
With the current resolution in the application of 25% (360x640) it would take up 170MB with full height and 25micron layer height. Lower resolution seems best while sliding and we can draw the higher quality picture (360x640) picture on a pause in sliding.

Rob2048, 2018-06-20:
I would recommend keeping the layers stored in the RLE format and only decoding them when needed. My realtime layer preview in javascript had no performance issues with this.

X3msnake, 2018-06-20:
Cool :) thanks for the insight

X3msnake, 2018-06-20:
@Rob2048 where is your voxel engine git, can't seem to find it in your account, What am I missing?

Rob2048, 2018-06-20:
@X3msnake I havn't put it up yet. Will do tonight :)

X3msnake, 2018-06-20:
Ok, add me to the project then ;)

Reonarudo, 2018-06-20:
@NardJ The name used on the 010 template is based on the name given to the field on their original software. I guess it makes sense to be variable depending on the resin.

NardJ, 2018-06-20:
@Reonarudo Looking in 010 Editor at for instance layerDef of layer #23, the variable for exposure time is still named "bottomExposureTime". I think this is an 'error' in the template. See this image:
[image]

If so I will rename the layer variable to 'Exposure Time' in PhotonFileEditor. Is this correct do you think?

Reonarudo, 2018-06-20:
Yes. I just checked its a mistake. Should be exposure time. Bottom exposure is defined on the file header. Probably a copy-paste error 😄

X3msnake, 2018-06-20:
these values have no effect on the file.

I can do some new testing tonight, but for what I remember from the tests i
did this does not apply, only the top level settings are used...

Like i said i will reconfirm this

NardJ, 2018-06-20:
I have implemented:
- Create new file
- Import bitmaps
- Export images from a photon file
- Replace image based on current layer/frame
- Allow up/down arrow to navigate layer / framesAllow setting and propagating a standart layer height (layers are set in absolute values it's a pain to manual edit)

Can anyone test these so we can strike these features as done?

Rob2048, 2018-06-20:
My slices to voxels code is up at https://github.com/Rob2048/PhotonTool

X3msnake, 2018-06-20:
@Rob2048 Awsome, love the unlicense license, true open source attitude :)

NardJ, 2018-06-24:
@X3msnake When applying new resin settings from a list, which settings are affected?

X3msnake, 2018-06-24:
Only exposure times since messing with the height will squash and deform
the model in z axis

X3msnake, 2018-06-24:
Exposure times and nr of bottom layers

No dia domingo, 24 de junho de 2018, Vinicius Silva <x3msnake@gmail.com>

NardJ, 2018-06-25:
@X3msnake: And per resin type do we have 1 exposure time for all bottom layers and one expore time for the other layers? Or can we have multiple exposure times across the bottom layers and also multiple for the other layers?

BTW we can split layers (reducing individual layer height) without effecting the model. Could this be benificial to resin settings?

X3msnake, 2018-06-25:
@NardJ 

The file structure ignores compleately the secondary exposure times. it even ignores the secondary off times so those values imo are just placeholders for future firmware.

I retested this and changing those values does not affect neither the bottom exposure per layer nor the normal layer exposure. so it is safe to think of those like unexistants for now. I would just set all those values to the same base/normal exposures 

as for reducing by doubling the layer images, it would be interesting yes since you can get more XY detail with some resins.

NardJ, 2018-06-25:
@X3msnake Does the Photon printer also ignore the secondary layer height? (If so I could hide the complete block with values)

X3msnake, 2018-06-25:
Layer height is used. We could design a variable layer from 2 files with
different slices.

Also the layers are in absolute distance. So a 0.05 layer will be in L1
0.05 on L2 0.10 L3 0.15...

And so on

X3msnake, 2018-10-11:
Most of these are already implementing, closing

closed 2018-10-11 by X3msnake