Photonsters/PhotonFileEditor #10: Bug: Import/encoding images not working

NardJ, 2018-06-20:
On import of images (Edit menu), the (png-)images have to be encoded in B/W RLE format. The encoded data is stored and slice 0 is displayed on screen. The displayed image has the right contours, but after each contour a random number of the (same) color is displayed. The encoded image does have the correct number of bytes. 

Probably after each color repetition to be encoded, the color value which is used in the encoded byte is the color of the first pixel which broke the repetition instead of the color of the repetition itself. If this is the case the code "encValue = color << 7 | nr" should be replaced by "encValue = prevcolor << 7 | nr"

X3msnake, 2018-06-20:
@Reonarudo maybe you can give some insights on this?

Reonarudo, 2018-06-20:
@X3msnake I'm having issues building the project to even be able to replicate the issue.

NardJ, 2018-06-20:
@Reonarudo: What kind of issues, which errors do you get? (Are you using python3?)

Reonarudo, 2018-06-20:
@NardJ Yes Python 3.6 through Anaconda + VSCode the issue might be the VCCode configuration. 
I was not being able to install pygame on the 3.6 instance but I have finally managed. Still when debugging no window shows up.

NardJ, 2018-06-20:
@Reonarudo And when not debugging, does it show up?

Reonarudo, 2018-06-20:
Yes 😄 ! it shows up when I run it directly from the terminal.

NardJ, 2018-06-20:
Nice! We should set up a wiki with this kind of info. I will look into it!

NardJ, 2018-06-20:
@Reonarudo For the wiki: Did you solve the problem with debugging?

Reonarudo, 2018-06-20:
@NardJ  No. Still no debugging. And I'm struggling a bit with the UI. My mouse clicks seem not to be at the cursor tip. I have to further investigate.

NardJ, 2018-06-20:
@Reonarudo What kind of system are you using? (Tested myself on 5 windows 10 laptops/desktops 32/64 bit)

Reonarudo, 2018-06-20:
@NardJ OSX 10.13.5 😄

NardJ, 2018-06-21:
If it is available, numpy is used for (a lot) faster decoding and encoding of pictures 
Closing and making new issue #28 for encounterd bug.

closed 2018-06-21 by NardJ