Photonsters/PhotonFileEditor #12: Decoding second preview image

NardJ, 2018-06-20:
When decoding the second preview image the same like the first I get an image which does not look right:
[image]

Is the color encoding really (like the first preview image) 2 bytes (little endian) with each bit like RRRRR GGG GG X BBBBB

X3msnake, 2018-06-20:
the image looks interlaced, isn't the size of this image smaller?

Rob2048, 2018-06-20:
The second preview is smaller and definitely follows the same encoding as the first preview.

Andoryuuta, 2018-06-20:
Yep, exactly what @Rob2048 said. 

The encoding is exactly the same except the size. The full preview is normally a 835x321 image and the preview thumbnail is normally 199x72 (Though you can override the filename text on the printer if you make it larger).

I can't look a the python code right the moment, but here is how I'm doing it in Go for reference:
https://github.com/Andoryuuta/photon/blob/master/preview_image.go#L9-L63

X3msnake, 2018-06-21:
[image]
[image]

X3msnake, 2018-06-21:
@NardJ These are taken from the .photon file I have uploaded to the resources to be the placeholder of newfile.photon ;)

Andoryuuta, 2018-06-21:
Aww, we're not going with cat photos? :P

[image]
[image]

But to be serious (not about the cats), that looks like a good default for new files. 👍

X3msnake, 2018-06-21:
@Andoryuuta Cool cats :P
I see you got the file editing figured out

@NardJ Master A has the solution :)

NardJ, 2018-06-21:
Yeah I think I know what the problem is. I am probably plotting the pixels of the 2nd thumbnail on a too large canvas/surface, hence the empty lines. Will check this tonight. Thx all!

Andoryuuta, 2018-06-21:
@NardJ If you don't mind a partial rewrite of `getPreviewBitmap` then PR https://github.com/NardJ/PhotonFileUtils/pull/23 should fix this issue.

No worries, if you don't like it / want to do something else though. :)

Rob2048, 2018-06-21:
Haha! Awesome!

X3msnake, 2018-06-21:
Awsome guys. thanks for your dedication and cordiality :)

closed 2018-06-21 by NardJ

X3msnake, 2018-06-21:
> Closed #12 <https://github.com/NardJ/PhotonFileUtils/issues/12> via #23

Perfect