Photonsters/PhotonFileEditor #13: Filling wiki

NardJ, 2018-06-20:
I have started a wiki to document possible  install problems and document functionality which is implemented. If you are missing info on the wiki, please feel free to add or reply here.

Reonarudo, 2018-06-20:
I'm quite new in Python.  Shouldn't we use a cross-platform project configuration/standard? Then it would be simpler for newcomers to deploy and run the code? [(Something like this for example.)](https://pip.pypa.io/en/stable/user_guide/#requirements-files)

NardJ, 2018-06-20:
Never used this, but from what I read this is definitely something we could use. Is this something you are willing to investigate?

Reonarudo, 2018-06-20:
Certainly. It also seems good practice to ensure repeatability of code deployment. I will create an issue focused solely on code deployment across all OSs

X3msnake, 2018-10-11:
I believe this is already implemented no?

If so can we close this?