Photonsters/PhotonFileEditor #14: Code deployment across Win/Linux/macOS

Reonarudo, 2018-06-20:
Prepare the project for easy deployment, both for development and execution.

X3msnake, 2018-06-21:
check the way @lexelby is doing it on inkstitch with travis, it's way out of my league but probably easy for you guys to understand

X3msnake, 2018-06-22:
https://anthony-tuininga.github.io/cx_Freeze/

cx_Freeze is a set of scripts and modules for freezing Python scripts into executables, in much the same way that py2exe and py2app do. Unlike these two tools, cx_Freeze is cross platform and should work on any platform that Python itself works on. It supports Python 2.7 or higher (including Python 3).

Reonarudo, 2018-06-22:
Seems very interesting. I'm currently testing Travis integration and will also test with such tools.

X3msnake, 2018-06-22:
http://cx-freeze.readthedocs.io/en/latest/distutils.html

X3msnake, 2018-06-22:
https://github.com/lexelby/inkstitch/issues/70#issuecomment-368345866

NardJ, 2018-06-25:
@Reonarudo I have added Travis CI as service.I am not familiair with Travis at all. Is this all I need to do, or do I have to fill in the form that follows?

[image]

Reonarudo, 2018-06-25:
After enabling Travis you have to activate it on a per-repository basis. You should get to a panel with a list of all your repositories with on/off switches. Then you just have to turn it on fo PhotonFileUtils.

NardJ, 2018-06-25:
@Reonarudo I think I have it activated now.

Reonarudo, 2018-06-25:
@NardJ I can't see it yet. Should be at: [https://travis-ci.org/NardJ/PhotonFileUtils](https://travis-ci.org/NardJ/PhotonFileUtils)

NardJ, 2018-06-25:
@Reonarudo I have the link working now (needed to give Travis permissions.)

Reonarudo, 2018-07-17:
This issue is not about generating an executable. It's about having a Continuous Integration system. I have been quite busy lately and didn't have time to merge the high amount of change done on the main branch with this one yet.

X3msnake, 2018-07-17:
Ah i see. Misread.

I shall open a installer issue then. Thanks

X3msnake, 2018-10-11:
@Reonarudo have you had time to look into this Master?