Photonsters/PhotonFileEditor #8: Export photon file to images

X3msnake, 2018-06-20:
With a photon file open have a export feature to export each layer to images as filename_#.png
Where # is the number of the layer

NardJ, 2018-06-20:
This is done (althought\ not yet uploaded to github.) Will keep this open until uploaded. 

Btw when do we close requests? After uploading (awaiting bug reports) or only after succesful testing?

X3msnake, 2018-06-20:
After testing. Usually with releases you can start a release and just add
the #nr of the issue in the description to do automatic issue closing and
keep a direct record of what release closed the issue.

You can also comment/close manually but we shouls always refer to the
update that fixed the problem

X3msnake, 2018-06-20:
The best way to test before closing is to create a test branch and merge
once it is confirmed

No dia quarta-feira, 20 de junho de 2018, Vinicius Silva <x3msnake@gmail.com>

NardJ, 2018-06-21:
Closing. Export is implemented. New request #22 asks to also include preview picture.

closed 2018-06-21 by NardJ