Photonsters/PhotonFileEditor #16: Creating new file crashes program

X3msnake, 2018-06-21:
[image]

X3msnake, 2018-06-21:
Fixed issue added template file to resources folder
also changed the colour of background to relate to the UV exposure light from the machine (What user sees when looking at machine without the vat and resin) Helps user to better understand what gets exposed and what does not get exposed

[image]
[image]

closed 2018-06-21 by X3msnake

X3msnake, 2018-06-21:
[image]
[image]
[image]

NardJ, 2018-06-21:
Great! Since I'm new to github: as I understand you made the changes to the master. Meanwhile the other branch was also editor. How do I pull these together... or does github do this automatically when I choose merge?