Photonsters/PhotonFileEditor #20: Allow typing numbers in the numpad

X3msnake, 2018-06-21:


NardJ, 2018-06-21:
Numpad keys are implemented, cycling field with enter/tab not yet.

X3msnake, 2018-06-21:
:)

NardJ, 2018-06-30:
@X3msnake I cannot reproduce this. I have .Windows 10 laptop which has a separate numpad and when numlock is on, I can enter numbers. Can you give me a walkthrough of what you do and how?

X3msnake, 2018-10-11:
@NardJ Still cannot use numpad for insertion of data Master, not even in the latest version, any ideas why?

When i press the numpad a commad is issued in the command window multiple times but no character appears
[image]

NardJ, 2018-10-24:
Weird, I just checked and with my laptop this works. It has a separate keypad. How is your keyboard layout?

If you modify PhotonEditor.py as below, PFE will also output the pressed keynumbers are pressed. This can help us debugging this further.

Around line 2873 you have the following comment
`                 #If numlock on then we use it to navigate layers`
Immediately after this insert
`                print ("Key:", isNumlockOn, event.key)`