Photonsters/PhotonFileEditor #24: Instalation step-by-step tutorial - wip

X3msnake, 2018-06-21:
Goto the project github and download the source zip file

[image]

[image]

---

Download and Install Python 3 from https://www.python.org/downloads/

[image]

[image]

---
From the windows menu Search cmd and run as admin

[image]

Confirm that your are running 3.0 or above by typing *python --version* into the command line

[image]

NardJ, 2018-06-25:
Nice! When ready we should add this to the wiki! (However not the readme I think) We should also add installation of numpy.

X3msnake, 2018-06-25:
@NardJ 

Yeah the idea is to add this to the wiki.

Readme should have only a context about what this is and can do, and links to the appropriate sections of the wiki

X3msnake, 2018-12-27:
Dependencies needed

Python must be on windows path!

python -m pip install -U pygame --user
python -m pip install -U opencv-python --user
python -m pip install -U numpy --user
python -m pip install -U PyOpenGL --user
python -m pip install -U scipy --user