Photonsters/PhotonFileEditor #29: Debate Forum

X3msnake, 2018-06-21:
This non issue can be used for discussion

X3msnake, 2018-06-21:
@NardJ when are you getting your photon? :)

NardJ, 2018-06-21:
Probably ordering this weekend from the warehouse in Germany. Just hope I have time to unpack with al things going on here ;-D

X3msnake, 2018-06-21:
Haha :D

Nice. Do you do 3D modelling? What is your call in getting yours?

Btw you are doing a great job. At this rate we will have a 1.0v ready for
release before we know it. Sorry I can't be much of a help where it comes
to coding :)

NardJ, 2018-06-21:
Will be fun to get it to work! I am a bit concerned about the smell of the resins, but I saw some solutions on FB to minimize it. Can't wait for the first succesful print, I probably will print a lot of miniatures to paint for D&D and other board games. 

I am not really a great CAD user and should make some time to learn a real CAD program. Currently I mostly model stuff to use in home improvement projects and model them with simple 3D modelling software like Tinkercad. I am also designing some GeoMag panels for someone using OpenScad. 

How do you use your Photon?

No problem with the coding, your help on the other aspects is great and you really keep the progress going! I never knew how much more comes to developing a new software program. (Btw. We should decide on the minimum functionality to make it in the version 1.0.)

X3msnake, 2018-06-22:
@Rob2048 are you in fb anycubic owners group?

People are leaving sugestions there for the voxel engine

[image]
[image]

X3msnake, 2018-06-22:
@NardJ 

There are resins with lesser smell than others. Harz Labs have good ones, nice smell too, not the fastest thing in the block but very usable.

I'm waiting on some ameralabs it's the fastest on the block according to the excel sheets samples i'll let you know how it is with the smell then.

You should try Fusion 360 it's free for Makers and Startups under 100K anual.

I have 2 Photons at the moment one at wokr for prototyping and one at home for research. My objective with the photon is to make parts for reparing broken stuff it's related to a project we have at our hackerspace besides the repair cafés that is calle junkchallenge (we teach how to disassemble, identify, repair recover or recycle) that's why i'm looking into metal casting with lost resin too...

For the minimum version we should open a issue or reconvert the one I already open with the lists ;)

Reonarudo, 2018-06-25:
@NardJ  Some images in the resources folder are missing on the last commit. Could you check if they are versioned?

Reonarudo, 2018-06-25:
@NardJ There are a few UI related image missing when trying to run directly from the current HEAD of the repo the first one it complains is cut.png.
`  File "/Users/reox/Documents/Git Repos/PhotonFileUtils/GUI.py", line 401, in __init__
    self.img=pygame.image.load(filename)
pygame.error: Couldn't open resources/cut.png`
I think you will be able to identify and fix it easily once you are home. 👍

NardJ, 2018-06-25:
@Reonarudo Sorry only see you comment now. Are you testing the New-features branch? I just updated this and all should be present.

Reonarudo, 2018-06-25:
@NardJ no I was running from master. Should I run from New-features?

NardJ, 2018-06-26:
@Reonarudo Yes there are a  lot of improvements and it should almost be ready to release as the new Master.

NardJ, 2018-06-27:
Well I just bought my own Photon... the price was low, even shipping from Germany, only €430!

NardJ, 2018-06-27:
Yeah, can I finally test the result of this Github! :-)

I just saw https://github.com/altLab/anycubic-photon-docs/ Very nice!

NardJ, 2018-07-03:
My photon has arrived! It is still in the box, because I will not have time for it the next couple of days due to the coding work I am on (the slicer). :-(

X3msnake, 2018-07-04:
@NardJ So have you done testing with the photon?