Photonsters/PhotonFileEditor #30: How are we going to integrate rob's voxel engine and when?

X3msnake, 2018-06-21:
Hey Masters, lets debate a bit on the best way to bring 3D visualization to the PFU.

Any ideas on how we can go about with it and timeframes for implementation?

X3msnake, 2018-06-21:
Have you added @Rob2048 to the team? I am sure he can help. 

What about integrating his engine as is with the gui maybe Rob can make an api that we can do calls from and to?

Rob2048, 2018-06-21:
I'll have some time to look at integrating it this weekend.

Rob2048, 2018-06-21:
@NardJ I think the code blocks 'without spaces/linebreaks' that you are taking about are in the three.js file? That is the OpenGL library I'm using, but all the voxel related code is within the index.html. It's actually a very small amount of code.

NardJ, 2018-06-22:
@Rob2048 I am currently cleaning up and annotating the code in the branche 'New-features'. I just did PhotonEditor,py. It is probably best/easiest to use this branch to integrate the engine in.

X3msnake, 2018-07-04:
@NardJ how's this comming along?

NardJ, 2018-07-05:
@X3msnake Code is cleaned up and engine can be ported into PFE. 

@Rob2048 If you need any help, let me know.

X3msnake, 2018-07-05:
@NardJ Awsome. Saw your tips on the loadscreen on phytonOpenGL

Can we set a ETA on this?

X3msnake, 2018-10-11:
@NardJ Does this still make sense now with the new internall voxel engine?