Photonsters/PhotonFileEditor #31: install numpy for faster import/export/navigating slice images

NardJ, 2018-06-22:
Since it isn't as clear in the closed issues: The newest version checks if the numpy library is available. If so it uses this library to decode/encode images. This makes faster exporting and importing of images possible. Also navigating through image layers and 'animating' the image stack by dragging the slider bar at the right edge of the layer image is possible.

X3msnake, 2018-06-22:
That means we should add instructions on how to check if numpy is installed
and how to install it if not. Correct?

NardJ, 2018-06-22:
Numpy is per default not installed (same as pygame), although numpy is one of the most common libaries.  I already updated the readme and wiki with how to install.

X3msnake, 2018-06-22:
Cool :)

Maybe add a alert sugestion in the cmd box if it does not find the numpy
lib?

X3msnake, 2018-06-22:
FYI 

If using anaconda instalation it already comes with numpy 1.14.3 installed

NardJ, 2018-06-23:
Thx, have updated readme and wiki (in the Master branch, should be ok I think)

closed 2018-06-23 by NardJ