Photonsters/PhotonFileEditor #32: Opening files with numpy crashes the program

X3msnake, 2018-06-22:
[image]

NardJ, 2018-06-22:
Did you update newfile.photon from the github?

NardJ, 2018-06-22:
Did a test myself and the newfile.photon in the Master Branch was not the correct one. After replacing it, it works.

NardJ, 2018-06-22:
Also had to update photonfile, there was a bug in between import and display of images.

X3msnake, 2018-06-22:
ok will try this soon and report back

X3msnake, 2018-06-22:
That's was not the problem.

Btw why did you change the .photon file I had made?

[image]

NardJ, 2018-06-22:
O I did not know you placed a photon file in there, sorry about that!
However I have a copy. I'll have a look why it is not loading properly.

X3msnake, 2018-06-22:
@NardJ no worries. btw these testing I have been doing is with the master branch.

NardJ, 2018-06-22:
Ok, this should do it. You template was (after decoding) a byte short. No idea, why but numpy-less decoding just stopped drawing and didn't care. The numpy decoding however tries to build a 1440x2560 matrix and crashes if the bytes don't match. I have now updated newfile with your version and inserted code to pad the decode image with bytes until it is correct.

X3msnake, 2018-06-22:
well the file was built with PhotonFileUtils so you better check that the
build algorithm is correct since it might be a bug in the non numpy code ;)

Just give it a double check :)

X3msnake, 2018-06-22:
Then that might be the reason why @Rob2048 dosen't show the last Layer in this one that was made with the PhotonUtils too
[LCD-UniformityTest-47u_Grid-50u_layers.zip](https://github.com/NardJ/PhotonFileUtils/files/2129305/LCD-UniformityTest-47u_Grid-50u_layers.zip)

X3msnake, 2018-06-23:
I don't know what you changed but now all built files are being saved wrong :(

X3msnake, 2018-06-23:
And what's with the reverse image?

NardJ, 2018-06-23:
I made the beginning for a test script (issue #35). (From now on I am strictly following this test script.)

The mirroring is fixed (just uploaded new file). I checked encoding an image with numpy and without and loaded both files with AnyCubic PhotonSlicer V1.3.6 and have no problem (see image below)

I also tried loading the file enclosed above (uniformity test) in PFE and resaving it. Both the original and the resaved load fine in Anycubic Photon Slicer. 

Can you confirm all is working correctly now? 
[image]

X3msnake, 2018-06-25:
What branch should i test?

NardJ, 2018-06-25:
@X3msnake Well now it would be best to test New-features. I think we can release this, but should test it myself first, before letting you encounter all sort of bugs. I will do a test tomorrow, but you are free to test also of course.

There is a pulldown box present to select Resin Presets, however at this moment it does not do anything.

X3msnake, 2018-06-25:
@NardJ XD

Hey don't be afraid of my bug finding skills :P

I can barely emulate the end user capability of screwing up simple instructions :D

closed 2018-06-30 by NardJ