Photonsters/PhotonFileEditor #35: Test script for new releases

NardJ, 2018-06-23:
Issue 32 shows we really need a test script to test new releases before merging them with Master branch.
Following items should be tested (please add/edit when needed)

- Does PFE start?
- Can we open a new file
- Open benchy.photon
- Replace bitmaps by smilie.png
- add number to all settings for a couple of layers
- Save photonfile to test.photon
- Load test.photon
- See if images and settings are the same.

- Load test.photon in AnyCubic Photon Slicer V1.3.6
- See if images and settings are the same.

Reonarudo, 2018-06-29:
Would be cool if the tests would be individual files per concern (File handling, Editor) and they could take advantage of `pytest` as it is used with Travis in the issue #14 branch

NardJ, 2018-06-29:
@Reonarudo Cool, but how does this work? Do you we run a py-file with a data file and expected returns values?