Photonsters/PhotonFileEditor #42: Disclaimer - Use at own risk

NardJ, 2018-06-26:
Yesterdag I found a bug in the New-features branch which generated layers at a lower position than the previous layer. This could make the Photon drive it resin-model back into the LCD display. 
Because of this I think it would be good to add an disclaimer to the project. What do you think?

> THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE OPEN GROUP BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.

X3msnake, 2018-06-26:
That's already in the License File you chose for the project no?

[image]

X3msnake, 2018-06-26:
But I think it is more important to warn users inside the software that they should do a dry run of the files generated with PFU

Maybe a "nag screen" on first boot of the program that you must click ok

something like
"
Reminder
- Use this at your own risk
- Backup your photon files before using the software
- Verify and dry test your files before doing a production run
"
And a checkbox to remove the nag screen
"I understand the risks stop nagging me"

NardJ, 2018-06-30:
Well a nag-screen is possible, however another (easier) option (to implement) is. Does this suffice?
[image]

closed 2018-07-03 by NardJ

X3msnake, 2018-07-03:
@NardJ This is a elegant and unintrusive solution Master, well done :)

NardJ, 2018-07-03:
Thx, I am also thinking about checking availability of Numpy and OpenGL and if absent a reminder to install them