Photonsters/PhotonFileEditor #43: Program crashes when access to folder is denied!

X3msnake, 2018-06-26:
[image]

Should instead popup a acess denied folder ignore the action and return to the program

NardJ, 2018-06-27:
Should be fixed now.

X3msnake, 2018-06-27:
Awsome Master
I'll hunt for more bugs ;)

NardJ, 2018-06-27:
@X3msnake, Happy hunting! (But I must admit I hope you don't find any in cut/copy/paste/deplicate, that was a hard piece of code to debug...)

X3msnake, 2018-06-28:
[image]

NardJ, 2018-06-30:
@X3msnake Should be working now. I made myself a windows directory and denied myself access to it. I could reproduce the error and now catched it so crash will not occur again. Please test if you have time (will leave this issue open, until I hear from you about this.)

X3msnake, 2018-07-04:
Working confirmed awsome!

commit ef6248b3 (2018-06-27): Fix #43 - crash if filedialog has no acces to directory
closed 2018-07-04 by X3msnake