Photonsters/PhotonFileEditor #45: What's the reasoning behind changing up/down behaviour? (NF)

X3msnake, 2018-06-26:
Dunno if it was intentional, but the up down button is switched from what it was on the MB.

For me it makes more sense as it was before up you go up one layer, unless you were starting the file from the top down.

My guess is that you align the up/down with the sliderbar, the problem is the bar should start in the bottom instead of the top since the layers start bottom up.

@NardJ 
Can you clarify your reasoning behing this?

NardJ, 2018-06-27:
@X3msnake My reasoning was that the model hangs upside down and so the layers build to the bottom. Futhermore it was easier to align the cursorkeys and navbuttons to the navslider than v.v. But we can switch them again if it is more intuitive.

X3msnake, 2018-06-27:
Ok makes sense. But from a user point of view it is easyer to expect that
up means plus. Down means less

This woul make sense if you were seeing the 3d model grow from the plate.
But we would not foece the user tob see upside down either.

What others think?

X3msnake, 2018-07-04:
@Andoryuuta @Antharon @Reonarudo @Rob2048