Photonsters/PhotonFileEditor #46: Use anycubic owners group resin table as base for resin times?

X3msnake, 2018-06-26:
There is a table curated by artus fuste and me that has good starting points for resins

https://docs.google.com/spreadsheets/d/1crvzMnt_8NJXAsABinoIhcOjE8l3h7s0L82Zlh1vkL8/edit#gid=0

maybe we can use this table to populate the software stock resin exposure times

NardJ, 2018-06-27:
@X3msnake The sheets is protected so I can not copy the data. Can you sent me an unprotected version?

NardJ, 2018-06-30:
@X3msnake Think you missed this one, but do you have an unprotected version and I have to ask but are we allowed to use it?

X3msnake, 2018-07-04:
[Anycubic Photon Resin Settings - Folha1.zip](https://github.com/NardJ/PhotonFileUtils/files/2164388/Anycubic.Photon.Resin.Settings.-.Folha1.zip)

NardJ, 2018-07-06:
Thx, I have implemented these in the fixes-and-features branch. If user tries to apply settings with different layer height, PFE asks for confirmation.

closed 2018-07-06 by NardJ