Photonsters/PhotonFileEditor #51: How to protect Master from commits

NardJ, 2018-06-27:
I am searching the web how to protect the master branch from (accidental) commits by myself. I want the commits to be to another development branch which will eventually be merged. However I can't find anything usefull.

I found the setting Branch Protection for master, however as I understand this only protects from merges and not from commits. Any ideas?

[image]

X3msnake, 2018-07-04:
Just activate Require Pull request reviews. You can override the pull request with your admin but you get a red warning

NardJ, 2018-07-05:
thx, turned it on!

closed 2018-08-13 by NardJ