Photonsters/PhotonFileEditor #52: STL loader

NardJ, 2018-06-27:
I am thinking about implementing a STL loader. Shouldn't be to hard to generate slicer.
Would this be a good thing or not something PhotonFileUtils should do? (There isn't any overlap with the Voxel Engine isn't there?)

Antharon, 2018-06-28:
not sure if good for photonFileUtils, but we already know how to set layer height. With combination of this we would be able to use adaptive layer height.

X3msnake, 2018-06-28:
Would be great :)

You think you can do it?

NardJ, 2018-06-29:
I figured out how to use OpenGL in python and it can be quite fast even with the old OpenGL 1.1 on my old laptop. Tested with 3DBenchy which has a lot of data points and it is acceptable. So we should be able to succesfully port Rob's 3d engine. 

Below a frist try at a rudimentary slice fo 3DBenchy. (proof of concept)

[image]

X3msnake, 2018-06-30:
Cool news :)

NardJ, 2018-06-30:
Well I sort of did it, see below. In the 3D viewer you can rotate and move the bed. 
The slicer is fully OpenGL and works, however thin edges are not always drawn and areas are not filled. Needs some more work.

[image]

[image]

NardJ, 2018-07-01:
Well solved this by calculating intersection from meshes. Now only to find a way to know which parts are inner parts and to floodfill.
[image]

NardJ, 2018-07-01:
Now developing a way to thicken inner walls instead of floodfill.

X3msnake, 2018-07-01:
@NardJ 

Awsome work Master, have you shared the code in a dev branch or somethig?

NardJ, 2018-07-02:
@X3msnake I have shared code in the branch STL-Loader. The code to thicken inner walls is almost ready and will be uploaded soon. I imagine this won't be the fastest code, so an optimization step for speed will follow.

X3msnake, 2018-07-02:
Awsome. After we add the ability to add manual supports and we have got
ourselves a slicer :P

X3msnake, 2018-07-02:
Found this engine and there is a open source to it, maybe we can drive some ideas from it

https://www.youtube.com/watch?v=nCDEOsgAt3g

NardJ, 2018-08-13:
Slicer is operational and produces slices at approx. 0.21sec/slice. (Slices are filled)

closed 2018-08-13 by NardJ