Photonsters/PhotonFileEditor #57: Preview Context image not working

X3msnake, 2018-06-29:
[image]

Also it would be great to be able to context import/export both preview images, at his moment on the full export you get only one of the images and should get the 2 images

NardJ, 2018-06-30:
Export of both previews is implemented. Import of previews is pending.

[image]

X3msnake, 2018-07-01:
@NardJ Awsome work master

Love the way you managed the disclaimer ;)

Ok both images are in fact exporting but they are exporting at the wrong size

Preview Zero is 360x186 and is ok
Preview One is  360x183 and should be 198x101

NardJ, 2018-07-02:
@X3msnake: The export size of the preview images was indeed wrong. Fixed this now, however... 

I find that all photon files have different sizes for preview0 and preview1. Is this correct that you know?
(I also checked with 010 Editor and customized template below.)

 
[anycubic-photon-hextemplate.txt](https://github.com/NardJ/PhotonFileUtils/files/2154088/anycubic-photon-hextemplate.txt)

X3msnake, 2018-07-02:
@NardJ

If you mean that the preview 1 is different from prev0 then this is correct all files will have diff sizes between these images

Since the smaller is used by the photon to show a preview along other files and the bigger one is shown when you select it.

If the size varies between files the it shouldnt happen and we may have something wrong in the decoding

NardJ, 2018-07-02:
@X3msnake A fault in decoding the image height and width (byte 6Ch and 70h for preview0) would result in a garbled decoding of the raw image data (bytes from 8Ch for preview 0). Also viewing several files with the (modified) 010 template show the same differences of size between several fresh Preview0 images. Could you verify you get the same different image sizes of the large preview images?

X3msnake, 2018-07-04:
@NardJ how can I verify this?

@Andoryuuta @Rob2048 @Reonarudo 

Do you have any knowlage why this might be happening?

Andoryuuta, 2018-07-05:
I'm not sure that I fully understand the issue. But I can verify that the preview image width, height and number of bytes it takes to store them varies depending on the file.

Here are the results from two different files:
```
reed.photon :
previewHeader Width:0x59c, Height:0x206, PreviewDataOffset:0x8c, PreviewDataSize:0x1780e, Field_10:0x0, Field_18:0x0
thumbnailHeader Width:0xc6, Height:0x45, PreviewDataOffset:0x178ba, PreviewDataSize:0x11c6, Field_10:0x0, Field_18:0x0

PHOTON.photon :
previewHeader Width:0x5a6, Height:0x399, PreviewDataOffset:0x8c, PreviewDataSize:0x645c4, Field_10:0x0, Field_18:0x0
thumbnailHeader Width:0xc6, Height:0x7f, PreviewDataOffset:0x64670, PreviewDataSize:0x33de, Field_10:0x0, Field_18:0x0
```

I never looked into how the slicer actually renders the initial QT image from the scene, but I know that it scales the preview thumbnail image right before it encodes it. (See the partially decompiled/pseudo-code here https://gist.github.com/Andoryuuta/010a64c283350e254a8fc1369dfb1396#file-previewencoding-cpp-L119-L148)

NardJ, 2018-07-05:
@Andoryuuta This is what we meant. I think we will have to test if PFE files with edited preview images load correctly on the photon. If so I will close this issue.

NardJ, 2018-07-06:
Just tested and preview images display garbled.

Andoryuuta, 2018-07-06:
@NardJ On which .photon file and which commit/branch/fork?

NardJ, 2018-07-07:
@Andoryuuta Branch fixes-and-features. I tested with smilie.photon where I replaced preview images with for instance the preview images from benchy.photon.

Reonarudo, 2018-07-08:
Do you have any info about the different sizes between the images? On my tests, there was no constant ration between them. Also is there any size limit like the screen size?

NardJ, 2018-07-08:
@Reonarudo, @Andoryuuta: I just did some testing. I used Smilie.photon as base file with the following results:
- Export preview 0 and replace preview0 with exported image -> Loads succesfully on Photon
- Resave exported image to png with GIMP. Replace preview0 with resaved image -> Loads succesfully on Photon
- Replace preview 0 and eventually also preview 1 with other images but of same size (see below). Found a bug where in the preview[1] data block the raw data address was not updated on replace of preview 0. Fixed this in the OpenGL branch (derivative of Fixed-and-Features). Result is this photon file: [Smilie._replaced543x401.photon.zip](https://github.com/NardJ/PhotonFileUtils/files/2174050/Smilie._replaced543x401.photon.zip) All addresses are fine in my opinion. Photon show preview image a little less garbled (less condensed to upper part of preview) but nonetheless still garbled. This makes me wonder if we are missing something about the compresseion of the images.

[image]
[image]

X3msnake, 2018-07-08:
Probably the nr pf colours. It must be a indexed 256 gif

X3msnake, 2018-07-08:
Something is really screwy here

I exported a image and then imported it again and the end value changed...

[image]

NardJ, 2018-07-09:
It seems the photon displays preview 1 (and not 0 as I assumed). On exporting preview 1 replacing preview 1 with this  exported version, the preview is not displayed correctly on the photon.

NardJ, 2018-07-09:
@X3msnake @Reonarudo @Andoryuuta It seems that the bytes which hold the nr of repeating pixels needed extra bits. The high byte has 4 high bits which are not needed for encoding the repeat-nr but should read 0011. 

Fixed this (in OpenGL branch) and tested replacing previews of smilie.photon with same resolution images. This works now. Further testing with replacing by different resolution images was also succesfull!

closed 2018-07-09 by NardJ

X3msnake, 2018-07-09:
Awsome :D

We can brand the example files that ship with the sw :P