Photonsters/PhotonFileEditor #59: Allow exporting context image!

X3msnake, 2018-07-01:
So i wanted to change the preview images only in a 700layer file. 

But i want to base the image on the original and do some modifications.

As we are now we need to do export bitmaps and that will take long while without any information as to where the process is hence why we need a progress bar #56

If we have a expost picture in the same area as import then we can export just the context image

NardJ, 2018-07-05:
@X3msnake Maybe I misunderstand you, but is this solved now we can replace preview images?

X3msnake, 2018-07-05:
Yes we can replace but we cannot export the preview only.

What we need is a way to export the image we are seeing, so we can edit it and replace it

NardJ, 2018-07-06:
Done (fixes-and-features branch)

closed 2018-07-06 by NardJ

X3msnake, 2018-07-06:
Your a beast Master :)

X3msnake, 2018-07-07:
> @X3msnake <https://github.com/X3msnake> Wait until you see the next
> functionality...OpenGL and the possibility to show 3d models or the Voxel
> Engine :-)

Coooool