Photonsters/PhotonFileEditor #60: Volume Calculator

X3msnake, 2018-07-09:
We urgently need a volume calculator for the .photon files or at least for the stl models

Ideally we could dial in the density and it would give us resin by weight, also we could change the "chicken" factor that would calculate extra resin and with a default of 10% extra resin 

https://github.com/mcanet/STL-Volume-Model-Calculator
https://github.com/tobob/simpleStlViewer

Reasons for this [Here](https://www.facebook.com/groups/AnycubicPhoton/permalink/1382138145264157/) and [Here](https://www.facebook.com/groups/AnycubicPhoton/permalink/1379063868904918/)

NardJ, 2018-07-09:
In the OpenGL branch in View menu there is a simple volume calculator. The calculation needs testing, the results seem too small, but should be right. Benchy.photon counts 2.0191.410 pixels and with 0.047mmx0.047mm per pixel and 0.05mm layerheight this calculates to 2.2ml

X3msnake, 2018-07-09:
Found this:

Meshmixer and the simple stl calculator must use the same algorithm and it is wrong. Bender 3D print plugin gives better results. if i had used blender i had not had a failed print :(

[image]
[image]
[image]

[image]

[image]

[image]

NardJ, 2018-07-09:
@X3msnake Seems like 150ml is the real volume. Will have to do some testing on my simple algorithm and see if correction is correlated to volume, surface area etc.

X3msnake, 2018-07-09:
@NardJ Yes 154ml actually since the file failed to print totally and i had poured 150ml into the vat mesured after degassing.

For what i've been reading the most accurate model is to fit the object with cubes so this might be easyer to do from the .photon file

NardJ, 2018-07-09:
@X3msnake Counting pixels and multiplying by 0,047x0.047xLayerHeight is exactly what I am doing. Can you post the photon file? I am curious if the calculation matches...

X3msnake, 2018-07-09:
here, show me the volume :P

[PHOTON FILE](https://github.com/NardJ/PhotonFileUtils/files/2177789/SB01P0035-SLA-lot.zip)
[STL](https://github.com/NardJ/PhotonFileUtils/files/2177810/____SB01P0035-SLA-lot-supported-full.zip)

X3msnake, 2018-07-09:
@nardj so? Suspense is killing me :)

NardJ, 2018-07-10:
Also around 120ml. I do calculate the pixels as being exactly 0.047x0.047xLayerHeight. Maybe the pixels project a little bit bigger (12% in each direction)? Will have to test other models and see if correction only depends on 2 dimensions. 
[image]

X3msnake, 2018-07-10:
That's interesting, tahnks.

All evidences point to resin contraction on curing then.

Need to run some more tests to debunk this

2018-07-10 13:27 GMT+01:00 Vinicius Silva <x3msnake@gmail.com>:

> Bo's Photon print validator also has a calculator but i can't load the
> file on it. it stops around layer 2290
>
> https://github.com/Photonsters/PhotonFileViewer/issues/5
>
>
>> Also around 120ml. I do calculate the pixels as being exactly
>> 0.047x0.047xLayerHeight. Maybe the pixels project a little bit bigger (12%
>> in each direction)? Will have to test other models and see if correction
>> only depends on 2 dimensions.
>> [image: image]
>> <https://user-images.githubusercontent.com/11459480/42490408-5372ae44-8410-11e8-84ec-ba8beabafa0e.png>
>>
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva

NardJ, 2018-07-24:
@X3msnake Not certain if I can close this one. Is the volume calc based on nr of pixels and pixel size accurate, or do we have to implement some kind of contraction correction ?

X3msnake, 2018-07-25:
Accurate thanks

closed 2018-07-25 by X3msnake