Photonsters/PhotonFileEditor #63: Algorithm needed to hollow model

NardJ, 2018-07-13:
I am in need of a good algorithm to hollow out models. Or to put it differently to make a model of the inner cavity. Approaches I am trying:
- displace point of original model in the inverse direction of the triangle normal
- place a sphere at the center and displace point of sphere outwards until minimal distance to points of model are reached.

Any ideas what a good but simple approach is? In my opnion the reason for hollowing is saving resin, so it does not have to be perfect, but speed is important.

X3msnake, 2018-07-13:
@Antharon @Reonarudo @Rob2048

X3msnake, 2018-07-13:
@NardJ  

Maybe it is easyer to do with the sliced model. take the pixel edge and count pixels in?

Or check Meshlad solution, it's open source so we can use it
http://www.meshlab.net/

NardJ, 2018-08-13:
Implemented in Ivan's branch with gyroid infill.

closed 2018-08-13 by NardJ