Photonsters/PhotonFileEditor #64: Add layer image editing

NardJ, 2018-07-15:
Make it possible to touch up layer images and possibly edit multiple layers at the same time.

NardJ, 2018-07-16:
@Photonsters/photon-file-editor Just implemented basic layer editing (coloring pixels) in the OpenGL branch. Just wondering what needs to be added to make this fully usable. Some ideas, but please add to them or strike them if unnecessary:
- Show lower layers as reference
- Color pixels from current layer to buttom layer (for making a support column)
- Multiple brushes, e.q. cylinder to bottom, small to large diameter towards bottom, draw angled to bottom
- Undo 

[image]

X3msnake, 2018-07-16:
Onion Skinning slider

[image]

NardJ, 2018-07-24:
Implemented in OpenGL2 branch. The shadow layers of lower layers do not scale properly yet and should have different color to distinguish from the current layer.

NardJ, 2018-07-24:
Now fully implemented in OpenGL2 branch. Loading shadow layers is still a bit to slow, will work better if parallel processing is used. (#66)

closed 2018-07-24 by NardJ