Photonsters/PhotonFileEditor #65: What is what in each branch?

X3msnake, 2018-07-16:
@NardJ 

I think it is time we have a official release, and we merge the new features all together in the master branche. I see we have a couple of branches but i can't fully grasp the differences between each one.

NardJ, 2018-07-16:
@X3msnake I have combined the Fixes-and-Features branch and the STLLoader branch into the OpenGL branch. The OpenGL branch will thus be merged to Master. 

I need to fix the following issues beforehand:
- Slicing models has a bug in which the area outside of the model-shell is filled.
- Direct editing op layer pixels is under development (couple of days work)
So I hope to do a merge this weekend or first half of next week.

X3msnake, 2018-07-16:
Awsome

NardJ, 2018-07-26:
Branch STFile en OpenGL are removed, since they are superceeded by OpenGL2

closed 2018-07-26 by NardJ

X3msnake, 2018-07-26:
Awsome