Photonsters/PhotonFileEditor #67: Printing time

NardJ, 2018-07-17:
Calculate estimated printing time.

NardJ, 2018-07-24:
Have implemented a rudimentary calculation: 
`nr_of_bottom_layers * ( off_time + exp_time_bottom)  + (nr_of_normal_layers  * ( off_time + exp_time)`

Is there a more accurate way? Do I need to account for other things?

Antharon, 2018-07-24:
In my opinion biggest problem with this calculation is, that people set off_time to zero by defaut, minimal off time is but 6,5s, which results in unrealistic print times

NardJ, 2018-07-24:
Is the minimal offtime enforced by the Photon printer?

Antharon, 2018-07-24:
6.5 seconds is time that printer needs to rise buildplate and dip it back into resin. If this process is prevented or peeling height is reduced, it might be little faster, but in general this is minimal off time in standard circumstances. Weird is, that it is counted into offtime.

NardJ, 2018-07-24:
Thx, corrected in OpenGL2 branch. (to be Master branch tomorrow)

X3msnake, 2018-07-24:
This is the correct way to do it

First the Peel Cycle must be configurable, either by a ini file like Bo did but preferably by a config menu in the software

here's the correct code;

```
var peel_cycle_time = peel_config_variable
var off_time = slicer_set_offtime

if (peel_cycle_time > off_time){
    total_off_time =  off_time + (off_time - peel_cycle_time)
} else {
    total_off_time = peel_cycle_time
}

total_build_time = nr_of_bottom_layers * ( total_off_time + exp_time_bottom) + (nr_of_normal_layers * ( total_off_time + exp_time)
````

NardJ, 2018-07-24:
Is this code really correct? Should the > not be a <? Now it reads:
* peeltime: 6.5s  & offtime: 4 sec ->  total_off_time: 1.5sec
* peeltime: 6.5s & offtime: 10 sec -> total_off_time: 6.5sec

I would expect the peeltime to be the minimum off-time and thus the code
`total_off_time = max (off_time, peel_cycle_time)`

X3msnake, 2018-07-25:
Yes you are correct. I meant to say offtime > peeltime

X3msnake, 2018-07-25:
Basically check if the off time is bigger than the config peel time, use peeltime if smaller or use offtime if bigger