Photonsters/PhotonFileEditor #69: Create installers / executables and use release

X3msnake, 2018-07-17:
We need installers or self contained executable releases for windows and mac at least.

As we stand the complexity of installation even tho it is well documented is a barrier for many users out there that want a turnkey solution.

if the sw is harder to install then use noone will get to see the value of it because most will not go the needed mile to install it :\

X3msnake, 2018-07-17:
Moved these links from #14

https://medium.com/dreamcatcher-its-blog/making-an-stand-alone-executable-from-a-python-script-using-pyinstaller-d1df9170e263

https://anthony-tuininga.github.io/cx_Freeze/

cx_Freeze is a set of scripts and modules for freezing Python scripts into executables, in much the same way that py2exe and py2app do. Unlike these two tools, cx_Freeze is cross platform and should work on any platform that Python itself works on. It supports Python 2.7 or higher (including Python 3).

https://regilanj.wordpress.com/2017/06/07/py-to-exe-in-python-3-6-1/

http://cx-freeze.readthedocs.io/en/latest/distutils.html

Found this:
https://www.reddit.com/r/Python/comments/5u15ab/creating_a_exe_file_for_a_python_36_script/

X3msnake, 2018-07-25:
@joaocarvalhoopen from altLab Collective managed to make a installer with cx_freeze

He shared the setup.py in the list, and i'm sharing it here in his behalf. this is for the Master branch, we may ned to add extra wheels libs for the other branches

@Reonarudo @NardJ 
With this config file we can probably automate travis to build exe releases for windows and the other OSs

[setup.zip](https://github.com/Photonsters/PhotonFileEditor/files/2227843/setup.zip)

X3msnake, 2018-07-25:
Some mor info on this

The setup.py works but you have to manually copy the software files to the genereated build for it to run.

There are ways to automate this, i'm trying to find it

Here's some refs
https://wiki.wxpython.org/cx_freeze

X3msnake, 2018-07-25:
Here's a working version of the setup that i did that creates a installer with an ico. just drop it in the same folder as PhotonEditor and run comand window as in comments.

[setup.py+ico.zip](https://github.com/Photonsters/PhotonFileEditor/files/2229494/setup.py%2Bico.zip)

X3msnake, 2018-07-26:
@NardJ 

I've just fixed this on this commit [c95b638]( https://github.com/Photonsters/PhotonFileEditor/commit/c95b638d6a223b7e00a8b1e772153ceb524f4ad2)

his script is already working and making a installer.

There are 2 issues to resolve.
1 - We need to find a way for the icon to get it's path automatically from the user selected install folder
2 - The files are being installed locked and that is a problem with the settings.ini

X3msnake, 2018-07-26:
@NardJ one solution to problem 1 is not to use the program files folder and use C:\instead

@bonosoft Any idea how we can fix this and still install in the programfiles folder?

NardJ, 2018-07-26:
Also numpy is not recognized.

X3msnake, 2018-07-26:
@NardJ Finally some code i can contribute to :P

X3msnake, 2018-07-26:
Got it working, uploaded the new changes just now to [7bae210](https://github.com/Photonsters/PhotonFileEditor/commit/7bae210bad2fdc87ba8eb7f380604787c720d5dc)

X3msnake, 2018-07-26:
@NardJ test the new PR and check if it is working for you too

X3msnake, 2018-07-26:
Awsome

[image]

What's theres wierd square characters in the bottom text?

NardJ, 2018-07-26:
Solved it, we needed to add numpy just like you did with opengl. Will upload new winsetup.py to OpenGL2 branch. This only leaves the two issues to be solved.
[image]

NardJ, 2018-07-26:
Ahm, we crossed paths... I overwritten your winsetup.py. Was your sollution the same?

NardJ, 2018-07-26:
Weird characters should be arrow symbols. Will fix this.

X3msnake, 2018-07-26:
My solution has numpy working ;)

X3msnake, 2018-07-26:
@NardJ what does Layer Height do?

It only changes the value of the layer height and does not recalculate the individual layer values!

NardJ, 2018-07-26:
The layer height in General does recalculate all layers. Maybe there is a bug in storing them. Opened issue #76 to check this.

NardJ, 2018-07-27:
@X3msnake I do not encounter issue 2 "The files are being installed locked and that is a problem with the settings.ini". I can change my preferred settings-view (basic->advanced) and after restart it was saved. Ideally however, settings.ini should not be part of the msi, since it is made after first start of PFE.

X3msnake, 2018-07-27:
I changed the default folder to c:/ root.

The issue occurs when you let it default to programfiles. Just try
commenting out the custom dorectory on c root and let the installer default
to program files

NardJ, 2018-07-27:
Well tried about everything to get an icon: 

A) To the last command 'setup (name=....' I added 'icon="photonsters.ico"' 
-> This gives the file PhotonEditor.Exe the correct icon.

B) In the shortcut_table I tried the following parameters for the icon:
"" -> default windows icon is used
"[TARGET]photonsters.ico" -> icon cannot be found
"[TARGET]\photonsters.ico" -> icon cannot be found
"TARGETDIRphotonsters.ico" -> icon cannot be found
"TARGETDIR\photonsters.ico" -> icon cannot be found
"DesktopFolderphotonsters.ico" -> icon cannot be found
"DesktopFolder\photonsters.ico" -> icon cannot be found
"[TARGETDIR]\PhotonEditor.exe" -> icon cannot be found
"PhotonEditor.exe" -> icon cannot be found
All prepend the icon file name with 
[image]

Only one I could get to work is (knowing the install directory beforehand):
"D:\PhotonFileEditor\photonsters.ico" -> icon is found

X3msnake, 2018-07-27:
I think we need to use Sys libraries with os.path.join

NardJ, 2018-07-27:
I think I fixed it, see snipets below. Can you try with new winsetup.py?

[image]

[image]

X3msnake, 2018-07-27:
Cool,

When you upload this tell me so I can test it on my side.

What about the locked edit permission, by deleting the ini file and letting
the sw create it does it get solved?

X3msnake, 2018-07-28:
@NardJ @Reonarudo 

Here's some info i got from Master @lexelby that hopefully will help us setup a travis CI with windows build.

> To be honest, I kind of blocked out a lot of what I figured out to get Windows going.  You can probably learn a lot from reading what I did in .travis.yml.
>  
> I'm not sure how much of it applies since you're using cx_Freeze and not pyinstaller like we use in Ink/Stitch.  At the heart of it though, the idea is to install Wine, and then install Python inside Wine's environment.  Then you run `pip install -r requirements.txt` to get everything your project needs, and finally run pyinstaller (or cx_Freeze).
>  
> Wine isn't a perfect Windows environment, but it seems to get the job done.  Travis doesn't have a true Windows build environment.
>  
> If you have any specific questions on why I did something a particular way in Ink/Stitch's .travis.yml, let me know and I'll try to explain.
>  
> Cheers,
> Lex

This is the yml that we are using to do the windows integration on inkstitch for windows
https://github.com/inkstitch/inkstitch/blob/master/.travis.yml

NardJ, 2018-07-28:
Replaced weird unicode arrow characters with images.
[image]

closed 2018-08-13 by NardJ