Photonsters/PhotonFileEditor #71: Add a fix my layer height button/action (floating point problem)

X3msnake, 2018-07-24:
[image]

Based on [this](https://www.facebook.com/groups/1194619320682708/admins/) and [this](https://www.facebook.com/groups/AnycubicPhoton/permalink/1397710053706966/) FB thread it would be great to have a quick fix button to stabilize the floater numbers intruduced by the slicer untill Chitu releases a fixed version of this issue.

NardJ, 2018-07-25:
If I understand correctly: The slicer does not round the layer height so it becomes something like 0.049999 instead op 0.05. This error accumulates since the cumulative layerheight is build adding 0.049999 to the previous layerheight instead of multiplying 0.05 to the layer Nr?

NardJ, 2018-07-26:
Is there a set op optimal layer height for the stepper motor in the Photon? In the docs section I see:

> MOTION DRIVER / MECHANICS
    T8 leadscrew with 2mm lead/pitch
    Nema 17 Motor 1.8º 200steps/rev
    Driver embeded Polulu Alegro 4998 set at 1/16 microstepping
    Hardware resolution 10µ (0.01mm)
    Digital microstepped resolution 0.625µ (0.000625mm)

Motor step is 0.01mm, so multiples of 0.01mm are valid and anything deviating from this will give artifacts in the prints?

NardJ, 2018-07-26:
I have implemented a check (tools menu in OpenGL2 branch) which checks if layerheight deviate from 0.01mm and a check if the cumulative layerheight deviates from layerNr*layerheight. After the check you can click ok to apply corrections.

X3msnake, 2018-07-26:
@NardJ 

If the value is 0.005 it will put the motor in strain but half step is doable.

closed 2018-07-26 by NardJ