Photonsters/PhotonFileEditor #76: Layer Height does not  recalculate the individual layer values!

NardJ, 2018-07-26:
@NardJ what does Layer Height do?It only changes the value of the layer height and does not recalculate the individual layer values!

NardJ, 2018-07-27:
@X3msnake Do you mean Layer height under Layer? 
[image]

These are the cumulative layer heights of each layer. I changed these values, stored the file and loaded it again. Layer height edits were saved. Can you give me more info to reproduce this?

X3msnake, 2018-07-27:
Ok so if you change the value by hand then it does change the values. But if you use the menu item it does nothing, here is a before and after hitting ok

[image]

[image]

X3msnake, 2018-07-27:
> The layer height tool checks if the layer heights diverge from the
> hardware resolution (0.01mm). In your file I see a Layer Height of 0.21mm
> which is a multiple of the hardware res. If the layer height was 0.202 it
> would make a correction to 0.20. Did I interpret/implement the hardware res
> wrong?
>
> *) Since all numbers in a photonfile are stored as 4 bytes, the accuracy
> of a float is precise and 0.30 will be stored as 0.29999999. I had an
> accepted error margin in the routine validateLayerHeight of 0.002mm and
> just reduced it to 0.001mm

so it is a machine file problem, not just the slicer?

The problem is that it did nothing even. let me check again with the new
updates you made.

NardJ, 2018-08-06:
Can we close this?

closed 2018-08-06 by X3msnake