Photonsters/PhotonFileEditor #80: Cython to speed up some code

NardJ, 2018-08-01:
Stumbled accross [Cython](https://en.wikipedia.org/wiki/Cython) which may have some use to speed up PhotonFileEditor.

NardJ, 2018-08-06:
Nice demo of inline Cython:

https://stackoverflow.com/questions/11731428/finding-first-non-zero-value-along-axis-of-a-sorted-two-dimensional-numpy-array

[image]

X3msnake, 2018-08-06:
> Nice demo of inline Cython:
>
> https://stackoverflow.com/questions/11731428/finding-
> first-non-zero-value-along-axis-of-a-sorted-two-dimensional-numpy-array
>
> [image: image]
> <https://user-images.githubusercontent.com/11459480/43713325-816d62c0-9979-11e8-8b91-a2dba01a4a55.png>

Question is can we make it work?

How would we go about. Does it need code rewriting

NardJ, 2018-08-06:
Well I think a lot can be achieved with smarter coding and parallel processing. This should be a last resort because it would python collaborators to help with programming.

NardJ, 2018-08-15:
Another [demo calculating mandelbroth images](https://github.com/cython/cython/wiki/examples-mandelbrot).

NardJ, 2018-08-15:
And from [docs.cython.org](http://docs.cython.org/en/latest/src/quickstart/build.html)
Building a Cython module using distutils¶

Imagine a simple “hello world” script in a file hello.pyx:
```
def say_hello_to(name):
    print("Hello %s!" % name)
```
The following could be a corresponding setup.py script:
```
from distutils.core import setup
from Cython.Build import cythonize

setup(name='Hello world app',
      ext_modules=cythonize("hello.pyx"))
```

To build, run python setup.py build_ext --inplace. Then simply start a Python session and do from hello import say_hello_to and use the imported function as you see fit.

One caveat if you use setuptools instead of distutils, the default action when running python setup.py install is to create a zipped egg file which will not work with cimport for pxd files when you try to use them from a dependent package. To prevent this, include zip_safe=False in the arguments to setup().

NardJ, 2018-08-16:
[Numba](http://numba.pydata.org/) is very easy to [implement](https://stackoverflow.com/questions/4641200/cython-inline-function-with-numpy-array-as-parameter), but not quite as fast as Cython...but very easy to implement
```
import numpy as np
from numba import autojit, jit

@autojit
def inc(arr, i, j):
    arr[i, j] += 1

@autojit
def test5(arr):
    for i in xrange(arr.shape[0]):
        for j in xrange(arr.shape[1]):
            inc(arr, i, j)
```
As the author on stackoverflow comments: 

> Even though it's 4.7x slower than Cython, most likely because the JIT compiler failed to inline inc(), I think it is AWESOME! All I needed to do is to add @autojit and didn't have to mess up the code with clumsy type declarations; 88x speedup for next to nothing!