Photonsters/PhotonFileEditor #81: Improve floodfill speed on slices

NardJ, 2018-08-01:
Some results on speed of floodfill using different libraries. (10 slices at 1mm interval of bunny.stl scaled to 40%, no parallel processing):

64 sec - Native python - stack based floodfill
25 sec - Pillow library
12 sec - Python-OpenCV library
 
[image]

X3msnake, 2018-08-01:
interesting.

OpenCv is what the monkey guy is using and it is a very powerfull opengl lib

X3msnake, 2018-08-01:
@NardJ Were can we see these tests?

NardJ, 2018-08-02:
On my pc at home :-) I am currently testing multiprocessing/-threading for a further speedup. After this I will start a new branch and upload it.

X3msnake, 2018-08-02:
JNC from altlab showed interest in looking into the optimization

NardJ, 2018-08-02:
Branch Slicer has latest code. Selecting full 3d view will automatically load bunny.stl. Pressing F5 slices the model at 0.1mm if I remember correct. 

The code is still a bit rough (not yet fully commented). Some optimization I did: (1) only floodfill if not yet filled; (2) parallel processing of slice and floodfill. 
(Sometimes a bug occurs when using floodfill in which case outside the slice perimeter is filled.)

Starting point for slicing is Slicer.py. This calls upon STLFile.py yo do the slicing itself.

NardJ, 2018-08-13:
Banch slicerv2 contains a vastly improved slice speed which uses floodfill a lot less. Speed is now 0.21 sec/layer.

closed 2018-08-13 by NardJ

X3msnake, 2018-10-09:
awsome work Master