Photonsters/PhotonFileEditor #83: implement vox import/export file format

X3msnake, 2018-08-07:
https://github.com/ephtracy/voxel-model/blob/master/MagicaVoxel-file-format-vox-extension.txt

if we implement this file format export we can open photon files in this open source voxel Photoshop and esit the photon sliced files :)

https://ephtracy.github.io

X3msnake, 2018-08-07:
Since MagicaVoxel is not Open source

Best alternative will be Goxel and VoxelShop

Voxelshop
https://www.youtube.com/watch?v=SQJiifHfZPE

Goxel
https://www.youtube.com/watch?v=h-52oe4P6jc

X3msnake, 2018-08-07:
list of 22 voxel editors

https://www.slant.co/topics/1544/~voxel-editors

X3msnake, 2018-08-07:
https://developer.nvidia.com/gvdb

X3msnake, 2018-08-07:
https://github.com/nickgildea/leven
http://ngildea.blogspot.com/
https://www.youtube.com/watch?time_continue=70&v=DkaxVzppDYU

X3msnake, 2018-08-07:
Voxel-Engine
A CPU based sparse voxel octree renderer.

https://github.com/bcmpinc/voxel-engine
https://bcmpinc.wordpress.com/

X3msnake, 2018-08-07:
Cross-platform, graphics API agnostic, "Bring Your Own Engine/Framework" style rendering library.

https://github.com/bkaradzic/bgfx

[image]

X3msnake, 2018-08-07:
https://github.com/Phyronnaz/VoxelPlugin

UNREAL ENGINE VOXEL PLUGIN

X3msnake, 2018-08-07:
GIGASPACE
http://gigavoxels.inrialpes.fr/download.html

   A CUDA/GLSL library for the efficient 3D real-time rendering of highly detailed scenes.

X3msnake, 2018-08-08:
https://github.com/armory3d/armory_docs
Armory is a game engine with full Blender integration, turning it into a complete game development tool.

Game Templates
Archviz Templates
Examples
Global Illumination
Voxel World
Material Vault

X3msnake, 2018-08-08:
Zmorph apparently uses Voxel thech in their slicer

https://voxelizer.com/

source: https://www.reddit.com/r/3Dprinting/comments/642nwq/voxelbased_file_formats/

X3msnake, 2018-08-08:
Monolith Is A Voxel-Based Modeling Engine For Multimaterial 3D Printing
http://www.monolith.zone/examples/

X3msnake, 2018-08-09:
https://medium.com/retronator-magazine/pixels-and-voxels-the-long-answer-5889ecc18190

X3msnake, 2018-08-09:
https://github.com/html5rocks/www.html5rocks.com/tree/master/content/tutorials/webgl/million_letters
https://www.html5rocks.com/en/tutorials/webgl/million_letters/

X3msnake, 2018-08-09:
﻿threeoctree.js (r60)
(sparse + dynamic) 3D spatial representation structure for fast searches
The aim of this project is to create a fully featured search tree for the THREE.js WebGL library.

This build is stable up to THREE.js ~r60
(updates to latest THREE build on hold as my time is required on other projects) 

https://github.com/collinhover/threeoctree

X3msnake, 2018-08-09:
WHAT IS VOLUME RENDERING?
Unlike traditional rendering uses triangles to display 3D graphics, volume rendering uses other methods such as Volume Ray Casting. This image-based method renders a 3D scalar field into a 2D image by casting rays along the 3D volume. Each pixel we see on the screen is the result of a ray that is going through the cube and getting intensity samples from the voxels at regular intervals.

http://www.lebarba.com

X3msnake, 2018-08-09:
Voxeltastic!
Ever need to ray trace your 3D images in some way that's more interested than slices? Don't want hard surfaces? Wanna do it in a web browser for the world to see? Try Voxeltastic! It's a A WebGL-based Raytraced Voxel Engine with Transparency. If you'd like a brief summary of what the default interface is, please click this link:

https://github.com/cnlohr/voxeltastic

X3msnake, 2018-08-09:
This is/will be a real-time OpenGL Sparse Voxel Oct-tree Raycaster. It makes heavy use of GPU compute.

http://danpburke.blogspot.com/2012/11/one-shiny-dragon.html

Voxel Rendering Test
https://www.youtube.com/watch?v=xW2YZBkk53I

SourceCode
https://github.com/burito/voxel

X3msnake, 2018-08-09:
VoxCad

VoxCAD (Voxel CAD) is a cross-platform open source voxel modeling and analyzing software. VoxCAD includes the Voxelyze 3D dynamics engine with large deformations, collision detection, multiple materials, volumetric actuation, etc. Check out the 

https://sites.google.com/site/voxcadproject/file-cabinet

X3msnake, 2018-08-09:
Voxelengine3 is a ThreeJS WebGL voxel engine. The code has NOT been polished and is provided "as is". There are a lot of code that are redundant and there are tons of improvements that can be made.

https://www.findbestopensource.com/product/lallassu-voxelengine3

X3msnake, 2018-08-09:
http://scholarworks.rit.edu/theses/9378/

Current technology allows MRI and other patient data to be translated into voxel-based 3D models for the purpose of visualization. However, these voxel models are extremely complex and are not suitable for rapid real-time manipulation. For true “on-the-fly” interactivity, polygon-based models must be hand-built using other methods and imported into a game engine.This project develops an algorithm to translate complex datasets into optimized models for real-time interactivity without sacrificing accuracy of the original imaging modality.

1.Osirix – Advanced Open-Source PACS Workstation DICOM Viewer http://www.osirix-viewer.com/

2. 3D Slicer- A multi-platform, free and open source software package for visualization and medical image computing http://www.3dslicer.org

3.MudBox – Autodesk¨ Mudbox™ 3D digital sculpting and digital painting software http://usa.autodesk. com

4.3D Coat – Retopologizing and 3D sculpting software http://3d-coat.com

5.MAYA - Autodesk¨Maya¨ 3D animation software delivers an end-to-end creative workflow with compression tools for animation, modeling, simulation, visual effects, rendering, matchmoving, and compositing on a highly extensible production platform http://usa.autodesk.com

6.Unreal Engine 3 – a complete development framework for PCs, iOS, Xbox 360¨, and PlayStation¨ 3, providing a vast array of core technologies, content creation tools and support infrastructure http://www. unrealengine.com/

X3msnake, 2018-08-09:
https://github.com/Gforcex/OpenGraphic/blob/master/README.md

OpenGraphic
Graphic Engine & Game Engine open source list！

NardJ, 2018-08-13:
What could help is, is erode the model to a shell width of 1 voxel and store the voxels as 3d coord's instead of storing complete voxel array.

NardJ, 2018-08-19:
@Photonsters/photonsters 
I almost have a voxel viewer in PFE, but I can't get my model to light properly. All sides have the same light intensity. The following is an octagon with a thickness, however the side walls have the same lighting intensity as the top walls. (Don't know about the light blue artifact line at the bottom)
[image]

My points array is a flat/1D array of triangle corner points. My normals array is a flat/1D array of normal 3d-vectors for each triangle. I feel like I am missing something really basic about using OpenGL. Any help is appreciated!

The following code is in the init procedure:

    # ===============================================
    # First add points to buffer
    # ===============================================
    points = numpy.array(points,dtype=numpy.float32)
    points=points * scale
    self.voxelPtBufferLen = len(points) // 3

    glEnableClientState(GL_VERTEX_ARRAY)
    self.voxelPtBufferIdx = glGenBuffers(1)
    dataSizePt = arrays.ArrayDatatype.arrayByteCount(points)

    # print ("datasize",dataSize)
    glBindBuffer(GL_ARRAY_BUFFER, self.voxelPtBufferIdx)
    glBufferData(GL_ARRAY_BUFFER, dataSizePt, points, GL_STATIC_DRAW)

    glDisableClientState(GL_VERTEX_ARRAY)

    # ===============================================
    # Second add normals for each triangle to buffer
    # ===============================================
    normals = numpy.array(normals, dtype=numpy.float32)
    self.voxelNmBufferLen = len(normals)

    glEnableClientState(GL_NORMAL_ARRAY)
    self.voxelNmBufferIdx = glGenBuffers(1)
    dataSizeNm = arrays.ArrayDatatype.arrayByteCount(normals)

    glBindBuffer(GL_ARRAY_BUFFER, self.voxelNmBufferIdx)
    glBufferData(GL_ARRAY_BUFFER, dataSizeNm, normals, GL_STATIC_DRAW)

    glDisableClientState(GL_NORMAL_ARRAY)

And in the loop I have the following code to draw the model

    colorFront = (0.1, 0.3, 0.6)
    colorBack = (1, 0.3, 0.3)
    glMaterialfv(GL_FRONT_AND_BACK, GL_DIFFUSE, (0.2, 0.2, 0.2))
    glMaterialfv(GL_FRONT_AND_BACK, GL_SPECULAR, (0.8, 0.8, 1.0))
    glMaterialfv(GL_FRONT_AND_BACK, GL_SHININESS, 50.0)

    glEnable(GL_DEPTH_TEST)
    glEnable(GL_NORMALIZE)

    glLight(GL_LIGHT1, GL_AMBIENT, (0.3, 0.0, 0.0, 1))
    glLight(GL_LIGHT1, GL_DIFFUSE, (0.6, 0.0, 0.0, 1))
    glLight(GL_LIGHT1, GL_SPECULAR, (0.9, 0.0, 0.0, 1))
    glLight(GL_LIGHT1, GL_POSITION, (10,10,10))
    glLight(GL_LIGHT1, GL_SPOT_DIRECTION, (-10,-10,-10))
    glEnable(GL_LIGHT1)

    glBegin(GL_QUADS)
    glColor3f(1.0,1.0,0)
    glVertex3f(10, 10, 10)
    glVertex3f(12, 12, 12)
    glVertex3f(10, 12, 12)
    glVertex3f(12, 10, 12)
    glEnd()

    glPolygonMode(GL_FRONT_AND_BACK, GL_FILL)
    glColor3fv(colorFront)

    # DRAW VOXELS
    glEnableClientState(GL_VERTEX_ARRAY)
    glBindBuffer(GL_ARRAY_BUFFER, self.voxelPtBufferIdx)
    glVertexPointer(3,GL_FLOAT,0,None)

    glEnableClientState(GL_NORMAL_ARRAY)
    glBindBuffer(GL_ARRAY_BUFFER, self.voxelNmBufferIdx)
    glNormalPointer(GL_FLOAT, 0, 0)

    glDrawArrays(GL_TRIANGLES, 0, self.voxelPtBufferLen)

    glDisableClientState(GL_VERTEX_ARRAY)
    glDisableClientState(GL_NORMAL_ARRAY)

X3msnake, 2018-08-20:
here's some more findings

https://github.com/topics/octree

@NardJ

look at the python octree one
https://github.com/inducer/boxtree

X3msnake, 2018-08-20:
https://youtu.be/gNZtx3ijjpo

X3msnake, 2018-08-20:
this one is for three.js
https://github.com/vanruesc/octree-helper/blob/master/README.md

X3msnake, 2018-08-22:
https://www.shapeways.com/forum/t/converting-voxel-data-to-supported-3d-format.1931/

NardJ, 2018-08-23:
Cool stuff. Maybe something for the next version. (First I am planning to complete the slicer and voxel viewer itself.)

X3msnake, 2018-08-28:
processing discourse octree library
https://github.com/thwegene/OCT/blob/master/README.md

X3msnake, 2018-09-10:
From Laser Scanning to Finite Element Analysis of Complex Buildings by
Using a Semi-Automatic Procedure

Cloud2FEM (Point Cloud 2 Voxel)

https://www.researchgate.net/profile/Alessandro_Lambertini/publication/280548230_From_Laser_Scanning_to_Finite_Element_Analysis_of_Complex_Buildings_by_Using_a_Semi-Automatic_Procedure/links/55b89ad708aed621de05eebb/From-Laser-Scanning-to-Finite-Element-Analysis-of-Complex-Buildings-by-Using-a-Semi-Automatic-Procedure.pdf?origin=publication_detail

X3msnake, 2018-09-10:
https://github.com/cagcoach/BlenderPlot/

A script for plotting voxel diagrams using Blender

Video Tutorial: https://www.youtube.com/watch?v=vrxsk6D1KhQ&feature=youtu.be

X3msnake, 2018-09-10:
https://www.paraview.org/overview/

ParaView is an open-source, multi-platform data analysis and visualization application. ParaView users can quickly build visualizations to analyze their data using qualitative and quantitative techniques.

Can read Voxel Data in XYZ format

https://www.youtube.com/watch?v=UwKFg0yOlCA&t=45s
https://www.youtube.com/watch?v=jbq26vWdNdE

X3msnake, 2018-09-10:
http://fiji.sc/

https://imagej.net/plugins/3d-viewer/index.html

[image]

X3msnake, 2018-09-15:
PAPER: Voxel-based Solid Models: Representation, Display and Geometric Analysis
https://pdfs.semanticscholar.org/b2ec/9d0c9fadba4174e1ae5d29fccb7b344dc365.pdf

X3msnake, 2019-01-08:
following [this](https://m.facebook.com/groups/1194619320682708?view=permalink&id=1549239798553990&refid=18&ref=m_notif&notif_t=group_comment_mention&__tn__=-R) thread in FB group here is another cat scan software to investigate

https://www.cti.gov.br/pt-br/invesalius/nova-versão-invesalius-31

X3msnake, 2019-01-08:
[OrtogOnBlender](https://www.blendernation.com/2019/01/08/ortogonblender-add-on-import-dicom-voxel-data-on-blender/?fbclid=IwAR3OE9F6tmNeUxPBQoBJPw5M2Ldy7kn3ZradVtTxGxDdrDXDj5XHrK6zmeY#prettyPhoto)
 add-on: Import DICOM voxel data on Blender

X3msnake, 2019-01-09:
[ITK snap, another CAT rebuilder](http://www.itksnap.org/pmwiki/pmwiki.php?n=Downloads.SNAP3&fbclid=IwAR2k5qarB-8kgorHJLAXXZpxfCldyOm2F9VjDXGVNhMEXdKT47filgNDvJQ)

NardJ, 2019-01-09:
[Voxel Engine in C++](https://stackoverflow.com/questions/48090782/how-to-best-write-a-voxel-engine-in-c-with-performance-in-mind)

NardJ, 2019-01-09:
[BinVox Editor](https://minecraft.gamepedia.com/Programs_and_editors/Binvox) could also be a good starting point. Some python code at https://github.com/dimatura/binvox-rw-py.

NardJ, 2019-01-12:
It seems that MagicaVoxel is limited to rendering volumes of 2048x2048x1024 (But the total number of solid voxels is still limited.) Goxel claims to be unlimited, but upon testing a large size it crashes. MagicaVoxel won't allow larger scenes (in the editor) of 126x126x126.

NardJ, 2019-01-13:
Did a test converting Photon file to STL file. It works, but takes a long time and generates large files (GBs). Slic3r could not load file due to memory issues and crashed. Best route for voxelviewer is raycaster.

X3msnake, 2019-01-13:
yes def raycast for visualizationand we need to compress the data with
octree and prpbably fractal compression. also would help to use only the
outside voxels of a model