Photonsters/PhotonFileEditor #85: PNG with Alpha Channel does not import / replace properly

thetrebor, 2018-09-12:
[image]
[image]

Steps to reproduce:
Import or replace a bitmap slice with a PNG image that has an alpha channel: notice that the image is imported as all-black.
Import or replace a bitmap slice with a PNG image that has NO alpha channel: notice it appears correctly.

Desired behavior:
Images import irregardless if PNG alpha channel is present.

NardJ, 2018-09-14:
Thx, will look into this!

thetrebor, 2018-09-14:
I looked at the code to see if could do it by nudging some arguments but, alas, it's beyond my skill.

krummrey, 2022-03-11:
I'm trying to get to load files from Photoshop back in. As I understand it the PNGs need to be 32bit (24 RGB and 8 Alpha). I used ImageMagik to convert 2 bit pngs to 32 bit:
`ls *.png | xargs -I% convert % png32:%`
The results seem to be correct:
`magick identify slice__0029.png`
`slice__0029.png PNG 1440x2560 1440x2560+0+0 8-bit sRGB 105243B 0.000u 0:00.000`
Still I can't reimport the images back in. 
`Stored: {'Action': 'replace', 'LayerNr': 0, 'LayerDef': {'Layer height (mm)': b'\xcd\xccL=', 'Exp. time (s)': b'\x00\x00HB', 'Off time (s)': b'\x00\x00\x00\x00', 'Image Address': b'\xd5\x1f\x00\x00', 'Data Length': b'\xe3\x81\x00\x00', 'padding': b'\x00\x00\x00\x00T\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, 'LayerData': {'Raw': b'}`
`…`
`}', 'EndOfLayer': b'\x1a'}} 4763823808 4763822400`
`order must be str, not int`

Any Idea how I can generate PNGs that I can reimport?

sn4k3, 2022-03-11:
Native sliced images are 24 bits, printers don't need the extra alpha byte.
You can use UVtools to do that kind of stuff, either via UI or scripting