Photonsters/PhotonFileEditor #94: Odd issue: Your images dimensions are are off and should be 1440x2560

thetrebor, 2018-11-06:
https://www.dropbox.com/s/2uumc4tp0r5af7q/Screenshot%202018-11-06%2018.17.40.png?dl=0
https://www.dropbox.com/s/jci0xs1vssavts8/Screenshot%202018-11-06%2018.18.18.png?dl=0

I'm confused because it looks like the images are the right orientation and size in any way i can determine but the program insists they are the wrong size (or maybe orientation?)

Returned:  /Users/Trebor/Desktop/Print Queue/!plates/bust-of-medusa-at-the-musei-capitolini-rome-1/1.png
   0 / /Users/Trebor/Desktop/Print Queue/!plates/bust-of-medusa-at-the-musei-capitolini-rome-1/1.png
Stored: {'Action': 'replace', 'LayerNr': 0, 'LayerDef': {'Layer height (mm)': b'\x00\x00\x00\x00', 'Exp. time (s)': b'\x00\x00HB', 'Off time (s)': b'\x00\x00\x80?', 'Image Address': b'&8\x0c\x00', 'Data Length': b'\xb1s\x00\x00', 'padding': b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, 'LayerData': {'Raw': b'...[[skip a few]]...}', 'EndOfLayer': b'\x1a'}} 48975401176 48803381112
Your image dimensions are off and should be 1440x2560

This is on the V2 branch
[image]

NardJ, 2018-11-07:
Hi,

The images to import should be in portrait orientation, so 1440 wilde and
2560 high. I have not het implementeren autorotation. Can you try what
happens if you rotate them nu hand before import?

Cheers, Nard

thetrebor, 2018-11-07:
(note, i just tried swapping the resolution x and resolution y in the general tool section and it had no affect)

NardJ, 2018-11-09:
Sorry for the late reply. I missed your message.
If I download your rotated image it's properties state it is 1439 x 2559. It has to be exactly 1440x2560

thetrebor, 2018-11-09:
You are correct, i when i was forcing it to be a certain pixel size i forgot to save the change.
Starting with a freshly sliced image and rotating it 90 degrees it imported but then i get this result:

[image]
[image]

I apologize for that bad test step.
We can either close this ticket and make a new one for "replaced image is all pink" or not.

I suppose the core issue is the program is working and it's just not giving me enough feedback to self-correct when i give it bad data.

X3msnake, 2018-11-09:
What are you using to produce the slices?

Are you trying to build a new file from PGNS or just add/edit a already
sliced photon file?

à(s) 16:29:

thetrebor, 2018-11-09:
I took an STL, loaded it into mango3d (a pre-release slicer) and used it to orient, support, hollow, and slice the file.  That gives me a zip of pngs.  I also exported the sliced/oriented model as an STL.
Then I sliced the supposed file in the photonic tool, saved it to get a good "receptacle" .photon file with the same characteristics. 
Then i took the zip stack from mango and ripped the alpha channel off all the files (because your tool hates on alpha channels for the moment)
Then I loaded the anycubic generated file into the PhotonFileEditor and tried to replace the images with the slices from mango.

That didn't work because of orientation issues, so then i tried just editing and importing one layer at a time just to understand what was failing.

So this build chain worked once, but then it stopped when I installed this all on a new computer.

It looks like ChiTu is getting better and handles hollowing and superior supports so this may not be necessary anymore as part of my workflow.  But there will be others, i presume, who will want to basically convert their slices zip into a photon file without going through anycubic tools.