Photonsters/PhotonFileEditor #95: Chitubox .Photon Generated file fails to load on Master branch

X3msnake, 2018-12-26:
Reports on failure to load .photon file generated from chitubox.

So far it fails in the Current release file we have available with a empty box with no text,

The master branch with the following error
[image]

Original FB thread [here](https://www.facebook.com/groups/AnycubicPhoton/permalink/1537984149679555/)

NardJ, 2018-12-26:
Tried to modify 010editor template to work for new file format, but no luck yet...

NardJ, 2019-01-11:
This is solved now in master and slicer branch. In Anycubic Slicer all data is continously. In ChiTuBox there are gaps, so when reading the data it is necessary to jump to prevAddr0, prevAddr1, layerDefAddr, etc.

@X3msnake: Do you know who build the windows setup? I don remember but don't think it was me. However I will look into making a build this weekend.

NardJ, 2019-01-11:
@X3msnake  I managed to create a new distributable and added it to the [previous release](https://github.com/Photonsters/PhotonFileEditor/releases/download/alpha/PhotonFileEditor-GL-alpha-0.1c-amd64.msi). On my pc it runs fine. The original source code for release 0.1b is gone, so I had to reconstruct it (see branch release).