Photonsters/PhotonFileEditor #96: Multiple windows Pop up!

X3msnake, 2018-12-27:
[image]

Reported by @rsilvers129

X3msnake, 2018-12-27:
@NardJ can you have a look at this, it does on import and on export

X3msnake, 2019-01-11:
@NardJ 

[image]

NardJ, 2019-01-12:
@X3msnake Import of images? Can you specify the actions you take, so I can reproduce this?

X3msnake, 2019-01-13:
new file, import images, choose folder, run, bam :)

NardJ, 2019-01-13:
@X3msnake Strange, I cannot reproduce this. Can you try with another image set?

X3msnake, 2019-01-13:
silvers reported the same issue, them again you have made changes will see
with the latest version if it happens