Photonsters/PhotonFileEditor #101: PhotonFileEditor Creates Corrupted Files

kawakawa77, 2019-06-03:
Hello There!

There seems to be a bug in Photon File Editor. So I was not able to save the changed Photon File, because Editor generates a corrupted Photon file.

How to reproduce:
I created a Photon-File of the GO-NOGO-Gauge STLs in ChiTuBox 1.4. Using Version2 Data (Speed-Settings) in Photon File too.

After that, I opened the Photon File in Photon File Editor ( (in master and in 10c-alpha too).

Then I saved the file without changes in File-Menu "Save As" Command.

The Editor created a corrupted file.

When opened the saved Photon File in Photon File Validator, it sayes "unknown format. Corrupted file".

Is there aa solution for this?

Regards
Artur

NardJ, 2019-08-25:
Hi, sorry for the very late reply. Since ChiTuBox the format is less tolerant. I have build a new PhotonFile reader/writer in Photonsters/Photonsters-FileEditor-v2 which should fix this. Hope to bring this out within a month.

bergstar, 2019-10-25:
Just wanted to confirm that this is still an open issue on Slicer-v2 branch. Tested today.
I also tested with files from Formware 3D - same error.

Is there any other way to use this software? Dose it work with any other slicer?