Photonsters/PhotonFileEditor #103: Strange behaviour with Elegoo Mars

sixtemesseven, 2019-08-25:
I tried inserting a png image into a template print to expose a PCB.

I tried this:
1. Make a project with elegoo mars in chiTuBox v1.5.0
2. Renaming it to .photon
3. Opened it in photon editor and replaced a layer with my 1440x2460 png file
4. Saved as .photon
5. renamed to cbddlp
6. Tried to print

The printer will start, but it will display a image from the last print job and not continioue with printing but it will get stuck...

This is my result when I replace layer 2 in the PE:
[image]

Onother thing is when I try to open the modified .photon with the PCB layer and saved as new .photon file once more in the photon editor I get this message:
Index -1 is out of bounds for axis 0 with size 0

Here are my .photon files which I experimented with:
[MyPrints.zip](https://github.com/Photonsters/PhotonFileEditor/files/3538166/MyPrints.zip)

Here is my png file:
[image]

NardJ, 2019-08-25:
Hi development of PhotonFileEditor (v1) is halted. I am currently fixing multiple issues in Photonsters/Photonsters-FileEditor-v2. 

With the lastest alpha version I made the attached photon file.
[ReplacePcbPngOnLayer0_NewFile.photon.zip](https://github.com/Photonsters/PhotonFileEditor/files/3538231/ReplacePcbPngOnLayer0_NewFile.photon.zip)
Can you check if this file works correctly?
(You should probably fix 'Exp. bottom (s)')

To reproduce: 
o Saved png file in your message above, 
o started PFEv2, select [File>New], 
o [Layers>Replace Image],  select saved png file
o [File>Save]

[image]

sixtemesseven, 2019-08-25:
Yes, this worked perfectly!
I was also able to run the source! Thanks!

closed 2019-08-25 by sixtemesseven