Photonsters/PhotonFileEditor #105: Incorrect output of the image on the UV display.

dark184, 2020-01-10:
Hello. There was a problem with PhotonFileEditor for WINDOWS. The latest version. Creating a PNG file 1440x2560 px. Uploading to PhotonFileEditor. Export (Save as) to a file *.photon. 
[image]
Instead of the expected image, a distorted image is displayed on the UV display. 
[image]
And on the touch screen of the printer, the picture is displayed correctly. Also, the image on the UV display is displayed with some magnification. What am I doing wrong?
[ANYCUBIC Photon.zip](https://github.com/Photonsters/PhotonFileEditor/files/4044809/ANYCUBIC.Photon.zip)
Printer ANYCUBIC Photon.

X3msnake, 2020-01-10:
you have the XY wrong rotate image by 90deg and try again

X3msnake, 2020-01-10:
or just export a image from a photon file an work over that image then
import again

dark184, 2020-01-10:
> у вас неправильное XY изображение, поверните его на 90 градусов и попробуйте снова. No dia sexta-feira, 10 de 2020, dark184 <notifications@github.com> escreveu:

Can you elaborate on how this is expressed? I also just read that all firmware versions except 4.0.11 and 4.2.18 have this problem. I have now firmware 4.2.3.

What can be associated with the increased size of the image on the UV display, with the firmware or uncorrected XY image?

X3msnake, 2020-01-10:
incorrect image orientation

dark184, 2020-01-10:
Honestly, I didn't understand. Probably a language barrier. PhotonFileEditor requires files that are only 1440x2560 pixels in size. If you rotate the image itself, it will not fit on the screen. If you rotate the entire image, PhotonFileEditor will not accept It.

X3msnake, 2020-01-10:
what version are you using?

Are you running the executable release or are you using python source?

X3msnake, 2020-01-10:
[TestMask.zip](https://github.com/Photonsters/PhotonFileEditor/files/4048555/TestMask.zip)

[image]

Looking again at your photo of the print it looks like frame shearing issues, part of your image is on the wrong place.

Try my file i just downloaded the release in a new win10 and made the image you shared and run it on my photon Classic with EPAX 4.2.18 firmware.

If your images does not look like my photo it is frame shearing.

As for the wrong size i presume you are saying the scale is wrong. You need to export your image from your gerber as 538dpi to match the resolution of the photon 47x47microns per pixel

ingobecker, 2020-10-28:
@dark184 have you managed to fix this problem?

X3msnake, 2022-10-11:
> 有没有一个大佬联系一下我，求这个程序教学，有偿

use UVTools, PFE is no longer being developed